bartoszbielawski/LEDMatrixDriver #7: How to

anouanou, 2017-07-09:
Hi guys

I saw this library on Arduino and I wanted to know how do you make it work with Arduino Uno for example.
Because I have the LedControl library which works perfectly but can't make this one work for me with all of the examples.

How should I connect my wires on the Arduino ?

"const uint8_t LEDMATRIX_CS_PIN = 9;"

I think that it doesn't work because of this line that I may have not configure properly.

Thank you

bartoszbielawski, 2017-07-09:
Hi,

From what I see it should be clock (SCK or CLK) to pin 13, DIN to pin 11
and then CS (SS) to wherever you choose (except for pin 10 and 12). 9
should be fine. Don't forget GND and 5V.

You can also try to lower SPI baudrate in the spi settings to something
lower (like few hundred kHz). The default I use for ESP 8266 is 5 MHz and
it could be too much for UNO.

Cheers,
Bartosz

anouanou, 2017-07-10:
Thank you it works perfectly now. Problem was that CS was connected to pin 10 which was forbidden.