bartoszbielawski/LEDMatrixDriver #17: Getting weird results from the examples

scottchiefbaker, 2019-01-28:
I picked up an [LED Matrix](https://www.ebay.com/itm/MAX7219-Blue-led-matrix-MCU-control-LED-Display-module-for-Arduino-Raspberry-nhj/192773943556?hash=item2ce238f104:g:rAoAAOSwqjhcJ0JE) from Ebay and I'm trying to use your library to interface with it. I've cabled it up per the specs and using the **Marquee Text** example I get the following output.

[image]

I'm assuming the dimensions should be 8x8 and four segments? When I use that only the last two segments light up. If I set it to 8x8 and two segments the above image is showed. The same text is shown on the left and right halves, and the left half is off by one pixel. I feel like I'm missing something really obvious, but I've tried everything I can think of.

bartoszbielawski, 2019-01-28:
The display you have is monochrome with 1 chip / segment? 
The explanation for the shifted half is easy - that part of display displays previous frame - because it thinks you have only 2 segments.
Did you check all the constants in the example? Line 11 (width, should be 32), line 13 (segments, should be 4). 
Try running some custom code that for example writes all pixels. If this one works I'll take a look at the example code and fix it. Let me know.

scottchiefbaker, 2019-01-28:
Oh that explains it... I just tried 32x8x4 and it works perfectly. I had it in my head it wanted the dimension of EACH segment.

closed 2019-01-28 by scottchiefbaker

bartoszbielawski, 2019-01-28:
I didn't write these examples, they were contributed by somebody else. 
In my own code I have just a single constant that is number of segments and I calculate everything based on that one.

Feel free to add comments or reorganize the code from examples if you have time :) Cheers :)