bartoszbielawski/LEDMatrixDriver #30: Bug in the "MarqueeTextClean.ino"

wEPac, 2019-03-16:
Hello, I think there is a little bug in this example when detecting a char width, line 320 :
`    int c_Width = 0;
    for(int c = 0; c < CHAR_WIDTH; c++)
    {
      if (font[asc][c] != 0)
        c_Width++;
}`
we are checking for an empty byte, starting from the least-significant bit. For a char like double quote ", this will truncate it. The right way would be to start from the highest bit and to check for the 1st no empty bit. It should be also faster ;)

bartoszbielawski, 2019-03-16:
I think I know what you mean, you suggest scanning chars from the end.
Maybe you could make a fix and a pull request? If not I can ask the original author of this example to improve it.

wEPac, 2019-03-16:
I will try the "pull request"... something new to learn for me lol

jamesgohl, 2019-03-16:
@bartoszbielawski please delete this example, it is highly inefficient.  Replace with my other one.

bartoszbielawski, 2019-03-16:
Okay, guys, then I close this bug report, I will remove the code in the future and replace it with a better version.

closed 2019-03-16 by bartoszbielawski