bartoszbielawski/LEDMatrixDriver #31: segmentation fault.

jamesgohl, 2019-03-16:
When I use flag "2" I get the following error:

C:\Users\Jam\Documents\Work\Arduino\MyProjects\MarqueeTextFast\MarqueeTextFast.ino: In function '_GLOBAL__sub_I_led':

C:\Users\Jam\Documents\Work\Arduino\MyProjects\MarqueeTextFast\MarqueeTextFast.ino:300:1: internal compiler error: Segmentation fault

 }

 ^

bartoszbielawski, 2019-03-16:
Oh, that's interesting... It's the compiler that sigsegvd... Any more
details?

jamesgohl, 2019-03-16:
C:\Users\Jam\Documents\Work\Arduino\MyProjects\MarqueeTextFast\MarqueeTextFast.ino: In function '_GLOBAL__sub_I_led':

C:\Users\Jam\Documents\Work\Arduino\MyProjects\MarqueeTextFast\MarqueeTextFast.ino:295:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

bartoszbielawski, 2019-03-16:
Can you tell me what's in that line? I don't have sources now.

jamesgohl, 2019-03-16:
I can't find '_GLOBAL__sub_I_led' anywhere.

295:1 ... } is the last brace in my example.ino file.

Here is the last function:

void writeNext()
{
  byte asc = text[textIndex] - 32;

  // We need to handle space character since ignoring whitespace.
  if (asc == 0)
  {
    scrollWhitespace = SPACE_WIDTH;
    // Force change to next character in text[].
    if (textIndex++ == textLen - 1)
    {
      textIndex = 0;
      scrollWhitespace = LEDMATRIX_WIDTH;
    }
    return;
  }

  // Not set for this character, so find start column.
  if (colLower == -1)
  {
    for (int c = 0; c < CHAR_WIDTH; c++)
    {
      if (font[asc][c] != 0)
      {
        colLower = c;
        colIndex = c;
        break;
      }
    }
  }

  // Not set for this character, so find finish column.
  if (colUpper == CHAR_WIDTH)
  {
    for (int c = CHAR_WIDTH - 1; c > -1; c--)
    {
      if (font[asc][c] != 0)
      {
        colUpper = c;
        break;
      }
    }
  }

  // Write the column to the right of the display.
  led.setColumn(LEDMATRIX_WIDTH - 1, font[asc][colIndex]);

  // Check if end of current character.
  if (colIndex++ == colUpper)
  {
    // Reset column start and finsh.
    colLower = -1; colUpper = CHAR_WIDTH;
    // Add whitespace.
    scrollWhitespace = CHAR_SPACING;

    // Check if end of text string.
    if (textIndex++ == textLen - 1)
    {
      textIndex = 0;
      scrollWhitespace = LEDMATRIX_WIDTH;
    }
  }
}

Regards,

James Gohl

*Are you cyber streetwise?*https://www.cyberaware.gov.uk/

bartoszbielawski, 2019-03-16:
The problem with this "bug" is that it doesn't actually happen in the library.
It's your compiler that critically fails (that is - crashes). 

Can you try to run the set pixel example? This may tell us if the problem is in your code or in the library itself.

jamesgohl, 2019-03-16:
I have removed the line that calls setColumn and it doesn't crash the
compiler... if that helps.

Regards,

James Gohl

*Are you cyber streetwise?*https://www.cyberaware.gov.uk/

bartoszbielawski, 2019-03-16:
Okay, let me look at setColumn.

I just made your example work. For ESP8266 it doesn't crash compiler.
1. Learn about forward declarations. (writeNext() is called from loop()
that is before writeNext()),
2. LEDMatrixDriver led(LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, 7);
This was my reason for class enums... You can write it like this but why
not:
flags = LEDMatrixDriver::INVERT_Y | ....? Then you don't need to explain
what does 7 mean.

BB

jamesgohl, 2019-03-16:
calling setPixel is working fine.

Within setColumn... something is wrong in the "value" part of this code
perhaps...?

        setPixel(x, y, value & 1);
        value >>= 1;

Regards,

James Gohl

*Are you cyber streetwise?*https://www.cyberaware.gov.uk/

jamesgohl, 2019-03-16:
okay so nothing has changed in setColumn since the last working version
which is really odd

Regards,

James Gohl

*Are you cyber streetwise?*https://www.cyberaware.gov.uk/

bartoszbielawski, 2019-03-16:
So before you didn't use this method?
You can try to replace it with:
```
setPixel(x, y, value & 1 ? true: false);
value >>= 1;
```
But as you have noticed yourself - there was no change.
We can write it directly in the framebuffer but it shouldn't change anything.

jamesgohl, 2019-03-16:
Ok also noticed something odd, the compiler crash is intermittent. Out of
10 compile and uploads, 5 of them failed with the crash, and they occur in
no particular pattern.  Perhaps my Uno is faulty, unfortunately I only have
uno Uno ;)

Regards,

James Gohl

*Are you cyber streetwise?*https://www.cyberaware.gov.uk/

jamesgohl, 2019-03-16:
.... but then again, this is compiled on my machine first.

Regards,

James Gohl

*Are you cyber streetwise?*https://www.cyberaware.gov.uk/

bartoszbielawski, 2019-03-16:
Maybe there was an update of the Arduino IDE? You can always try to compile
it online using this Arduino Create.
Or some other editor - VSC + PlatformIO or something...

jamesgohl, 2019-03-16:
Updated arduino ide, didn't seem to update gcc though and made no
difference in the crash results.

I reverted back to my version of the driver which *was* working fine for
every compile and now I'm getting the same crash results.  How very odd.

I am going to guess the problem is at my end, why would I get intermitted
compiler results unless my ram is somehow corrupt?

Regards,

James Gohl

*Are you cyber streetwise?*https://www.cyberaware.gov.uk/

jamesgohl, 2019-03-16:
I can't reproduce the crash after retarting my machine.

jamesgohl, 2019-03-16:
> 1. Learn about forward declarations. (writeNext() is called from loop() that is before writeNext()),

I know about forward declarations, yes anything that is called needs to be declared before, but gcc doesn't seem to care.

closed 2019-03-16 by jamesgohl

bartoszbielawski, 2019-03-16:
I'm pretty sure it's the Arduino editor that hides it from you. My compiler
(gcc for Tensilica Xtensa) certainly cares.

BB