bartoszbielawski/LEDMatrixDriver #35: Invalid initialization of reference of type 'Adafruit_GFX&' from expression of type 'LEDMatrixDriver'

kd8bxp, 2019-06-06:
Board: Wemos D1 Mini (ESP8266)
IDE: 1.8.9
OS: Linux Mint 19.1
Adafruit GFX Library version 1.5.3
I'm getting the above error when trying to compile the AFGFXMarquee example.  
`/tmp/arduino_modified_sketch_83714/AFGFXMarquee.ino: In function 'void loop()':
AFGFXMarquee:76:63: error: invalid initialization of reference of type 'Adafruit_GFX&' from expression of type 'LEDMatrixDriver'
   copyBitmap(canvas, cntr, 0,lmd, LEDMATRIX_WIDTH - 1, 0, 1, 8);
                                                               ^
In file included from /tmp/arduino_modified_sketch_83714/AFGFXMarquee.ino:18:0:
/home/lfmiller/Arduino/libraries/Adafruit_GFX_Library/afbuffer.h:15:6: note: in passing argument 4 of 'void copyBitmap(GFXcanvas1Read&, int16_t, int16_t, Adafruit_GFX&, int16_t, int16_t, uint16_t, uint16_t)'
 void copyBitmap(GFXcanvas1Read& src, int16_t sx, int16_t sy,`

I put the afbuffer.h and afbuffer.cpp in the Adafruit_GFX_Library directory.  That didn't work.
I moved the afbuffer.h and afbuffer.cpp file to the LEDMatrixDriver directory, that didn't work.
I tried to switch the board to the UNO and got this error
libraries/Adafruit_GFX_Library/afbuffer.h:4:19: fatal error: cstdint: No such file or directory
compilation terminated.

Not sure if this issue should be here or on the AFBuffer repository, So hopefully the right person sees it, and can tell me what I've done or how to fix it.

bartoszbielawski, 2019-06-06:
Hi,

Sorry for the late reply, I'm traveling and I have no access to a computer.

Adafruit GFX is an optional dependency. By default it's disabled and that's
why the compiler doesn't recognize lmd object as a valid reference.

Check the line here:
https://github.com/bartoszbielawski/LEDMatrixDriver/blob/dfaeee893f8f40544068c2a7b68e2cd5e838f92d/src/LEDMatrixDriver.hpp#L28

Enabling it by adding the right #define should solve the problem.

Can you make a bug report for the other problem in the other project? The
header is supposed to define *int*_t types but it's probably not available
on all platforms.

I hope this helps,
Bartosz

>

bartoszbielawski, 2019-06-10:
I'm back and I have access to my computes. Did my solution work for you?

bartoszbielawski, 2019-06-11:
I have added instructions in the readme on how to enable the optional dependency. 
Let me know if you still have the problem.

closed 2019-06-11 by bartoszbielawski