bartoszbielawski/LEDMatrixDriver #36: calling clear() after setEnabled(false);

jamesgohl, 2019-06-21:
Whilst the led matrix controller is in disabled state after calling `led.setEnabled(false);`, trying to enable it again cleanly seems impossible.  

So when I have try:

`led.clear();
led.setEnabled(true);`
or
`led.setEnabled(true);
led.clear();`

The issue is, when enabling again there is a sligh flash of the previous text on the display, the order in calling the two functions above doesn't make a difference.

It could be my controller and/or limitation of the hardware but I am wondering if there anything which can be implemented in this driver to clear the driver buffer and the device buffer whilst in the disabled state?

bartoszbielawski, 2019-06-21:
Hi James, 

Enabling/disabling display doesn't clear the memory in the chip. Calling `lcd.clear()` clears only the local frame buffer. 

I believe the right order of operations is:
`
lcd.setEnabled(false);
lcd.clear();
lcd.display();
lcd.setEnabled(true);
`

Does it solve your problem? 

If I were to change the behaviour, what would be the best way to do it? The assumption was from the very beginning that it's the user who decides when to flush to the hardware. `setEnabled()` maps directly to the hardware command so it would be a bit confusing if it started doing something else as well...

jamesgohl, 2019-06-22:
Hi Bartosz

What I am doing is already what you are describing in that very order 😊

Yes I wouldn’t be comfortable having the clear command actually clear the
hw at the same time, perhaps another command to flush the hardware?

J

bartoszbielawski, 2019-06-22:
>  Yes I wouldn’t be comfortable having the clear command actually clear the hw at the same time, perhaps another command to flush the hardware?

Hmm, that's what `display()` does?

I have an idea but tell me yours so I can check if what I came up with makes sense :)

jamesgohl, 2019-06-22:
Okay that has given me another idea, let me try calling display() after
clear() on the driver buffer first.

bartoszbielawski, 2019-06-24:
Did your idea work?

jamesgohl, 2019-06-24:
It did improve slightly, there was some flickering on the display showing
the previous leds for a noticeable, but the duration was significantly
shorter.
What else can I do?
So I tried, clear(), display() setEnable(true)

bartoszbielawski, 2019-06-24:
Let me get this straight, you want to disable display, clear it and then enable it and see it empty, without flicker of the previous content, right?

If 'setEnabled(false); clear(); display(); setEnabled(true);' still causes flickering then I will have to look into the problem. Maybe I don't understand how is the display sleep mode handled.

jamesgohl, 2019-06-24:
Yes I am basically using setEnable how I think should be used, to quickly
enable and disable on-the-fly.
Perhaps a clearDisplay() function needs adding which does both but handles
better.

bartoszbielawski, 2019-06-24:
Can you make a video of the problem happening + the code you use? I will
try to check it at home.

The manual specifies: The display drivercan be programmed while in shutdown
mode, andshutdown mode can be overridden by the display-testfunction.

So I see no reason why it would blink while being powered up.

bartoszbielawski, 2019-06-25:
I didn't manage to reproduce the problem. If I clear the display when it's disabled I just shows nothing when I enable it. Can you send me your code and maybe a recording of the problem?

closed 2019-07-04 by bartoszbielawski

jamesgohl, 2019-07-05:
> I didn't manage to reproduce the problem. If I clear the display when it's disabled I just shows nothing when I enable it. Can you send me your code and maybe a recording of the problem?

It could be just the chip I am using.

Sorry I haven't had the time to play with this, I will create a recording and send you this weekend.  The code I am using is interconnected with a windows application which sends data over serial to the arduino.

bartoszbielawski, 2019-07-05:
No hurry, if it's a bug, it's not critical ;)
You can send me your code as well. Maybe I can spot the problem...