bartoszbielawski/LEDMatrixDriver #37: More scroll options

jamesgohl, 2019-07-05:
`void LEDMatrixDriver::scroll( scrollDirection direction )` is very useful, but everything is scrolling away and gone forever.

Might I suggest an optional parameter to the scroll function which "wraps" the lost pixels back to the other side of the matrix display, so it appears continuous.  I think this is quite a low-level feature which would open up the driver to have more potential for animations.

bartoszbielawski, 2019-07-05:
Let me check how complex would it be...

jamesgohl, 2019-07-05:
Yes direction and detecting the lost byte after shift, but you know which
bytes will be discarded and can store them in a temp var before the shift.

The complications arise when you have to take into account the x/y/segment
flips ;)

J

bartoszbielawski, 2019-07-05:
> Yes direction and detecting the lost byte after shift, but you know which
> bytes will be discarded and can store them in a temp var before the shift.
>

These are bytes for vertical scroll and bits for horizontal one.

>
> The complications arise when you have to take into account the x/y/segment
> flips ;)

I think it doesn't matter, really.￼ The memory layout is always the same,
it's the display method that handles flips.
That's why I wanted it done this way :)

>

bartoszbielawski, 2019-07-07:
Do you have time to test the code? I have pushed a new branch: wrapped-scroll, it contains the changes. It works on my machine ;)

jamesgohl, 2019-07-10:
Will see if I can tomorrow evening I am swamped with projects at the moment
my friend.  Thank you for responding and implementing so quickly <3

bartoszbielawski, 2019-07-10:
Take your time, I'm not in a hurry :)

jamesgohl, 2019-07-11:
My 7219 boards have stopped working together, must have damaged the copper
pate when I removed pins before joining them together.  I’ve ordered some
more from China.

bartoszbielawski, 2019-07-12:
> My 7219 boards have stopped working together, must have damaged the copper pate when I removed pins before joining them together. I’ve ordered some more from China.

Bad luck... You can try to patch them up, remove the damaged segment. 
It will be weeks before new ones arrive... I will be on holidays by then ;)

jamesgohl, 2019-07-12:
Yes they will 😂 I can run in 4 segments for now to test.  As you can see,
a mess.  There is no copper plating under the solder 😂

bartoszbielawski, 2019-07-12:
> Yes they will 😂 I can run in 4 segments for now to test. As you can see,
> a mess. There is no copper plating under the solder 😂
>

I have a test setup with hand-soldered 5x1 display :)  It's useful to check
if I didn't make a mistake rounding somewhere to the nearest power of 2 ;)

jamesgohl, 2019-07-12:
Good call! 

Testing this now, just while I remember though...

`void scroll( scrollDirection direction, bool wrap=true);`

Default false, would be the prefered, since the feature didn't exist previously.  But I know you know that :)

jamesgohl, 2019-07-12:
Okay this is me writing to confirm that it is working nicely :)

bartoszbielawski, 2019-07-12:
Great! Then I fix the header (I wanted it to be false, why did I put true?)
and merge the code.

jamesgohl, 2019-07-12:
Great :) You're only human :D (I hope so)
Okay I'm thinking the driver could maybe do with a tiny example with some chevrons <<<< scrolling just to expose the feature.

bartoszbielawski, 2019-07-12:
I already made an example with some scrolling stripes. Maybe <>s are
better...

closed 2019-07-17 by bartoszbielawski