bartoszbielawski/LEDMatrixDriver #40: integrate AFBuffer into this library

iZhangHui, 2019-07-24:
It won't be neccessay to install AFBuffer library.  

 For one previous project, I used [Adafruit_SSD1306](https://github.com/adafruit/Adafruit_SSD1306), which supply simple APIs to display strings and bitmap.

I look forward to this library becoming more powerful!

bartoszbielawski, 2019-07-24:
The idea was to be able to have a hardware independent frame buffer to render some things in advance. 
You can use AFBuffer with SSD1306, you render something in the memory and then copy it to the hardware. I used it this way here: https://github.com/bartoszbielawski/InfoClock32/blob/master/src/displayTask.cpp#L94

Is there any other reason why I should merge the two libraries?

iZhangHui, 2019-07-24:
@bartoszbielawski Thanks for your explaination.

I think that class LEDMatrixDriver inherits from GFXcanvas1 may be better.
```cpp
class LEDMatrixDriver: public GFXcanvas1
```

bartoszbielawski, 2019-07-24:
Inheriting from GFXCanvas1 forces us to use the internally malloc-allocated buffer. My code allows us to use an external buffer. GFXCanvas1 inherits from the Adafruit_GFX anyway...

closed 2019-09-20 by bartoszbielawski