bartoszbielawski/LEDMatrixDriver #41: Scroll Left to Right

malltinaa, 2020-03-07:
Hi,
Thanks for sharing this library.
I use the [MarqueeText.ino](https://github.com/bartoszbielawski/LEDMatrixDriver/blob/master/examples/MarqueeText/MarqueeText.ino) example.
How do I scroll from left to right. (I use Arabic.)
I made a few changes but I didn't succeed.

Thanks.

bartoszbielawski, 2020-03-07:
Hi,

This example has been created by one of the contributors and they basically created all the logic with right-to-left in mind. The complex solution is to try to invert all the loops (drawing characters, drawing on the display).
The simple solution is to invert the whole display in x axis, then pixel column 0 will be on the right. 
To invert the display pass INVERT_DISPLAY_X in the constructor of the object:
https://github.com/bartoszbielawski/LEDMatrixDriver/blob/master/examples/MarqueeText/MarqueeText.ino#L15
Use:
LEDMatrixDriver lmd(LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, INVERT_DISPLAY_X);
Of course latin letters will be inverted but I guess you need to provide your own letters anyway.
Let me know if this helped.
Bartosz

malltinaa, 2020-03-07:
Thanks for the quick reply
But I get the following error:
```
MarqueeText: 15: 59: error: 'INVERT_DISPLAY_X' was not declared in this scope
 LEDMatrixDriver lmd (LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, INVERT_DISPLAY_X);
                                                                                                              ^
exit status 1
'INVERT_DISPLAY_X' was not declared in this scope
```

bartoszbielawski, 2020-03-07:
My bad. I forgot the constant lives in the class. Try this:

`LEDMatrixDriver lmd (LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, LEDMatrixDriver::INVERT_DISPLAY_X);`

malltinaa, 2020-03-07:
> My bad. I forgot the constant lives in the class. Try this:
> 
> `LEDMatrixDriver lmd (LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, LEDMatrixDriver::INVERT_DISPLAY_X);`

I tested. Unfortunately this method does not work.

bartoszbielawski, 2020-03-09:
Well, if you can tell me what doesn't work I can suggest a solution. You can always write your own code from scratch. The library doesn't really care about texts, it cares only about pixels.

malltinaa, 2020-03-10:
> Well, if you can tell me what doesn't work I can suggest a solution. You can always write your own code from scratch. The library doesn't really care about texts, it cares only about pixels.

Thanks Bartosz,
I used the [original code](https://github.com/bartoszbielawski/LEDMatrixDriver/blob/master/examples/MarqueeText/MarqueeText.ino).
According to you, I made the following change to the code and output was the same as the video below:
`LEDMatrixDriver lmd (LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, LEDMatrixDriver::INVERT_DISPLAY_X);`

[Demo](https://vimeo.com/user109454851/review/396634431/3743a157e6)

bartoszbielawski, 2020-03-11:
I'm sorry for late replies, for some reason I get no notifications.
Also the vimeo link doesn't work...
If I find time I may try myself at home later today.

malltinaa, 2020-03-11:
> I'm sorry for late replies, for some reason I get no notifications.
> Also the vimeo link doesn't work...
> If I find time I may try myself at home later today.
Thanks,
I changed the video link.

bartoszbielawski, 2020-03-11:
Ah, I see the problem. Now segments are inverted as well. Try this one:
LEDMatrixDriver lmd (LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, LEDMatrixDriver::INVERT_DISPLAY_X | LEDMatrixDriver::INVERT_SEGMENT_X);

And let me know if it helped.

bartoszbielawski, 2020-03-11:
You can also try to modify this example:
https://github.com/bartoszbielawski/LEDMatrixDriver/blob/master/examples/AFGFXMarquee/AFGFXMarquee.ino

It uses a different approach. First it renders text into a buffer and then only copies it to the display.
It may be that making this one work needs only inverting a counter or two.

idreamsi, 2020-03-12:
> Ah, I see the problem. Now segments are inverted as well. Try this one:
> LEDMatrixDriver lmd (LEDMATRIX_SEGMENTS, LEDMATRIX_CS_PIN, LEDMatrixDriver::INVERT_DISPLAY_X | LEDMatrixDriver::INVERT_SEGMENT_X);
> 
> And let me know if it helped.

Thanks, Everything is okay but apparently the text is mirrored. Please see the [video](https://youtu.be/HwhTLAk4-s8).

> You can also try to modify this example:
> https://github.com/bartoszbielawski/LEDMatrixDriver/blob/master/examples/AFGFXMarquee/AFGFXMarquee.ino
>
> It uses a different approach. First it renders text [...]

I'll examine this example. Of course, because I made the font, I need a function like `setpixel(x, y)` to display.

idreamsi, 2020-03-12:
I solved the problem.
Replace lines [110 to 112](https://github.com/bartoszbielawski/LEDMatrixDriver/blob/master/examples/MarqueeText/MarqueeText.ino#L110-L112) with the following code:
```
  #ifdef L2R
  //Scroll left to right
    if( ++x > LEDMATRIX_WIDTH ){
    x = -len*8;
    }
  #endif
  #ifdef R2L
  //Scroll right to left
    // Advance to next coordinate
  if( --x < len * -8 ) {
    x = LEDMATRIX_WIDTH;
  }
  #endif 
```
Only un-comment one of the lines below. (Insert these lines at the beginning of the code.)
```
#define L2R
//#define R2L
```

bartoszbielawski, 2020-03-12:
Would you care to make a pull request with your changes? Some other people could benefit from the improved example.

idreamsi, 2020-03-12:
> Would you care to make a pull request with your changes? Some other people could benefit from the improved example.

I will do.

bartoszbielawski, 2020-03-12:
Thank you. Can I close the issue or do you have more questions?

closed 2020-03-12 by malltinaa