bartoszbielawski/LEDMatrixDriver #42: Custom characters for 7 segments modules ( did an alternative wiring of the 7 segments)

Belamat, 2020-03-11:
Hello Bartosz, my name Is Matías, I've made a custom 7 segment module with the max7219 chip, your library is the only one that I could use out of the box, with another SPI module (the Nrf24L01).

The thing it that of course, since I wire the 7 segment in an alternative way, the numbers and characters, dont work nice in my case, im trying to solve it in a simple way, how could I do it? Im an intermediate/beginner programmer (never did my own libraries and very novice to low level programming)

I've attached a diagram of the custom wiring and a photo of the real circuit i've made.

(the custom wiring i did was just to make a single layer pcb test)

Any help/guide will be very well welcome thanks for your contribution and support!

[image]

[image]

bartoszbielawski, 2020-03-11:
Okay, I see the problem. So basically you had to swap segment pins to make it easier to cable.
You can't use the built in digit decoder in the chip but you can still use it in "raw" mode.
Try it like this first (I'm not compiling the code, please correct is as needed):
https://gist.github.com/bartoszbielawski/143dee460015ef41cb8250072d506686

The first piece of code (till the end of the for loop) should light a single different segment on all the displays. Now you have to "design" your own digits and put the in the digitMap array and then - it's kind of done.

I may modify the interface to be able to set a row in one go using nicer interface than the getFrameBuffer(). Let's see if this works first.

BB

Belamat, 2020-03-11:
#include <LEDMatrixDriver.hpp>

const int NO_OF_DRIVERS = 1;

LEDMatrixDriver lmd(1, A2);

void setup() {lmd.setEnabled(true);
  lmd.setIntensity(2);  // 0 = min, 15 = max
  lmd.setScanLimit(7);  // 0-7: Show 1-8 digits. Beware of currenct restrictions for 1-3 digits! See datasheet.
  lmd.setDecode(0xFF);  // Enable "BCD Type B" decoding for all digits.
  // no setup needed
}
void loop() {
  
uint8_t* digits = lmd.getFrameBuffer();
for (int  i = 0; i < 8; i++){
    digits[i] = 1 << i;
}
lmd.display();

uint8_t digitMap[] = {B11110000};

digits[0] = digitMap[0];  //display zero at 0th position

}

couldnt get a character, probably doing something wrong (got many mixed ones turned on and off)

Belamat, 2020-03-11:
//couldnt get a character, probably doing something wrong (got many mixed ones //turned on and off)

#include <LEDMatrixDriver.hpp>

const int NO_OF_DRIVERS = 1;

LEDMatrixDriver lmd(1, A2);

void setup() {lmd.setEnabled(true);
lmd.setIntensity(2); // 0 = min, 15 = max
lmd.setScanLimit(7); // 0-7: Show 1-8 digits. Beware of currenct restrictions for 1-3 digits! See datasheet.
lmd.setDecode(0xFF); // Enable "BCD Type B" decoding for all digits.
// no setup needed
}
void loop() {

uint8_t* digits = lmd.getFrameBuffer();
for (int i = 0; i < 8; i++){
digits[i] = 1 << i;
}
lmd.display();

uint8_t digitMap[] = {B11110000};

digits[0] = digitMap[0]; //display zero at 0th position

}

Belamat, 2020-03-11:
got something now, think i had to remove the 

md.setDecode(0xFF); 

now i got just one led of each 7 segments turned on

bartoszbielawski, 2020-03-11:
Remove setScanLimit() (you already have 8 displays, that's the default) and setDecode(0xFF);
Exactly!

bartoszbielawski, 2020-03-11:
Now the loop is for checking which segment is which. Now make binary numbers that correspond to digits you want to display. Then remove the loop and only use the part with digitMap.
Should be fairly easy.

Belamat, 2020-03-11:
Ok, check that, and how do I then use the digits, for example if I want to show the number 3456, how do I call that? Not shure how to link the digit map with the normal way o calling numbers

bartoszbielawski, 2020-03-11:
Once you have uint8_t digitMap[] = {0xAA, 0xBB, 0xCC...};

Display on the first fur digits the number.
digits[0] = digitMap[3];
digits[1] = digitMap[4];
digits[2] = digitMap[5];
digits[3] = digitMap[6];
lmd.display();

In general you will have to extract units, tens, hundredths and send them to the display.
You'd need to do exactly the same having regular 7seg as it never handles long numbers, just single digits.

Belamat, 2020-03-11:
[image]

Like a Champ!!

Thanks a lot Bartosz!! It's been a pleasure! I'll keep on working now and then I'll see how it goes. You are great! Thanks : D !!

bartoszbielawski, 2020-03-11:
Great! What is this device exactly?

I have just checked the API and you can (and should) actually use setDigit(segment, value) method instead of the one with the pointer.

Belamat, 2020-03-11:
Its a project I'm working on, its a device that registers heart pulses via RF and then indicates a sets a cycle of  breaths depending on the count of pulses ,(synchronizing heart beats and breath) indicating each moment by sound (audio clips) and light (neopixelmatrix)  . Once finished it will be shared : )

Belamat, 2020-03-11:
Not shure how to use the setDigit function. I get **'setDigit' was not declared in this scope**

Calling it in this way:
**setDigit(2, B11010001);**

bartoszbielawski, 2020-03-11:
It's a method so call it on the object. lmd.setDigit(...);

Belamat, 2020-03-11:
Great! Works also! Is there a way here in GitHub to give you some credit for helping? The stars?

bartoszbielawski, 2020-03-11:
Stars are for you to observe the project. This one is quite stable, so don't expect many updated ;)
 If you ever publish your project feel free to link to this repo. 
How do you measure heartbeat by using RF? You meant the NRF2401?

bartoszbielawski, 2020-03-11:
Ah, I know. If you had time, you could strip your code and I could include it as an example for "swapped" 7seg displays. You know, fork, add code, make a pull request. This, of course, would be credited to you.

closed 2020-03-11 by bartoszbielawski

Belamat, 2020-03-12:
Yes sure Bartosz! Be glad to share and credit everyone who added to the project in one or another way. : )

I measure pulse with another device that sends via NRF24L01 to the main device (the one of the photo).