bartoszbielawski/LEDMatrixDriver #46: Can't get more than 4 modules to work

wachl, 2021-05-17:
Hi,
I sucessfully use this library to address one PCB with 4 modules on it. However when I daisy chain another pcb to it (with seperate power ofc) I can't get it to work. I am using the set pixel example with `LEDMATRIX_CS_PIN = 15;` and `const int LEDMATRIX_SEGMENTS = 8;` on an ESP32.

What am I doing wrong?

Thanks!
`

bartoszbielawski, 2021-05-17:
Hi,

> However when I daisy chain another pcb to it (with seperate power ofc) I can't get it to work.

Hmm, what is so obvious about using separate power? I have been using the library myself with 8 segment displays and the power was also "chained". In reality it's not chained, segments are connected in parallel. 

Maybe you swapped some pins? Maybe the power is not connected as it should be?
How it "doesn't work"? Only 4 displays work? None of them? Some blinking/glitches?

>  I am using the set pixel example with `LEDMATRIX_CS_PIN = 15;` and `const int LEDMATRIX_SEGMENTS = 8;` on an ESP32.

This looks okay. I'd look at the connection first, check continuity, check if signals reach the last module.

Cheers,
Bartosz

wachl, 2021-05-17:
Wow you are fast! Thank you very much!
I just wanted to close the issue, I had a defective unit :((

Now everything is working as intended!
Thanks for this awesome library!

bartoszbielawski, 2021-05-17:
> Now everything is working as intended!

I'm glad it wasn't my code ;)

> Thanks for this awesome library!

It's nice that people use it, enjoy!