bartoszbielawski/LEDMatrixDriver #47: Flickering lines when multiple leds are lit

wachl, 2021-05-17:
Hey,
I'm back with another issue (that's probably my fault):

When I light up multiple leds lines start to randomly flicker (as seen in the video). I tried multiple things which had little to no effect so far:
- changing the main loop delay()
```
  void loop() {
  button.check();
  lmd.display(); //flush display
  delay(10);
}
```
- removing `lmd.display();` from the main loop
- only using 4 8x8 modules

Have you experienced such issues? Could I try to change the SPI speed or do you think it is a power issue?

Thanks a lot!

bartoszbielawski, 2021-05-17:
I'd test it like this:
1. slow down the SPI,
2. clear display and display it when it's empty,
3. turn on LEDs one by one and check how far you can get.

wachl, 2021-05-17:
could you hint me in the right direction: I can not find anything on how to change the SPI speed on an ESP32...
Thanks!

bartoszbielawski, 2021-05-17:
https://github.com/bartoszbielawski/LEDMatrixDriver/blob/cd14cdd4f3a2cc7d7789a798f7b97ee8b202f2b9/src/LEDMatrixDriver.cpp#L16

bartoszbielawski, 2021-05-17:
Let me know if the issue is solved. 
I was actually considering making an API that does not force SPI speed and just leaves it alone, maybe that could be a reason to do it.

wachl, 2021-05-17:
Thanks for the tip, SPI speed has no influence as far as I can tell (although I only tested it shortly). I will try to find out how many leds I can drive without issue and report back!
Thanks for your help!

bartoszbielawski, 2021-05-26:
I will be closing the issue if there's no proof that the library is at fault.

bartoszbielawski, 2021-05-26:
It's okay, I can wait, take your time.
Some people after solving their problems never come back to even leave a comment...
Cheers!

wachl, 2021-05-26:
Alright so I tried it out with [this library](https://github.com/squix78/MAX7219LedMatrix) and interestingly the issue shows as well. I guess this is good news for you and your library but bad news for me :) However I found out that the flickering only occurs at low intensity levels (less than half of the max brightness) which leads me to believe it is a hardware issue of the MAX7219 chip (maybe because supposedly the cheap ones are fake). I will try to dim the panels using the r_set resistor value.

Thanks for your support!

wachl, 2021-05-26:
Update: Dimming using Rset works without flickering! On my boards a 20k resistor is used and I am now running 200k atm without issues. I'd say brighness is comparable to software intensity = 0

bartoszbielawski, 2021-05-26:
Thanks for letting me know. Did you try to get a better power supply? Maybe 5V was dipping...

wachl, 2021-05-30:
I have tried different power supplies without success. But since I found out that flickering is not occuring on full brightness I ruled out power delivery problems. Also I think you can see in the video, that the flickering lines are not dimming but rather lighting up (which is in line with the vanishing of the flickering on full brightness).
Seeing as this is a hardware problem (possibly MAX2719 clone problem) I am fine with closing the issue.
Thank you for your patience and support!

bartoszbielawski, 2021-06-01:
Thanks for letting me know, cheers!

closed 2021-06-01 by bartoszbielawski